Emilv2/huawei_solar #91: Can't connect to SUN2000-6KTL-L1 + SDongleA-05

webdados, 2022-07-02:
Hi guys,

I bought the SDongleA-05 and installed on my SUN2000-6KTL-L1, latest firmware versions installed for both. Dedicated IP attributed to the Dongle (can be pinged from the internal network). Modbus enabled and slave set to 1 as explained here: https://github.com/Emilv2/huawei_solar

Then I installed the integration on my home Assistant, tried to connect with correct IP, port 6607 and Slave 1 with no success.

Now I notice this information on the integration installation:

> What IP-address and port should I enter?
> Starting from firmware updates released in December 2021, Huawei has closed the Modbus-TCP interface on the network to which the inverter connects. ie. If the inverter is connected to your home network on 192.168.1.11, it will no longer be possible to connect on that IP. This also applies for connecting via the Dongle. You will need to connect to the inverter using the WiFi AP.
> 
> To configure your WiFi access, connect to the WiFi SSID SUN2000-<inverter serial number>, and use the Modbus-TCP interface available on 192.168.200.1. In most cases, the port has been moved to 6607 instead of 502.
> 
> How do I connect to the inverter WiFi?
> There are multiple possible approaches which are discussed on our Connecting to the inverter Wiki-page

How have you guys solved this? The whole idea of haveing the dongle is connecting to Modbus. My Home Assistant needs to be on my network, not on the inverter network.

Anyone got this working?

webdados, 2022-07-02:
That quote was from another Huawei Solar integration, not this one, but I think the information still applies, as I now have installed this integration and I'm getting the same exact problems.

webdados, 2022-08-02:
I can now get the data from the dongle, after updating the inverter to PC124 and the dongle to PC130.

I'm now stuck on which attributes to use on the "Energy" section of Home Assistant.

[image]
[image]

shagadeli, 2022-08-12:
Any chance you could share the firmware files? I can't really find a way to download them from Huawei.

webdados, 2022-08-12:
Sure: https://we.tl/t-qvMdmK3jtP

webdados, 2022-08-12:
Issue solved.

closed 2022-08-12 by webdados

krzakx, 2022-10-02:
Hi guys.
I am just trying to runthis on my config wihtout succes.
Just updateded my Dongle to V100R001C00SPC130 and inverter to V100R001C00SPC146 added snipcode to configuration.yaml

```
sensor:
  - platform: huawei_solar   
    host: '192.168.x.xx'
    optimizers: false
    battery: false
    slave: 1
    port: 6607
```
Verified configuration.yamls in HA Developer section but I can't see any new entities. What I need to do? 
@webdados do I need seperate WiFi dongle or not? How is your hardware set-up at the end?

webdados, 2022-10-03:
@krzakx I'm now using this integration and all is setup in the GUI: https://github.com/wlcrs/huawei_solar
I'm connecting to the dongle via ethernet.