Emilv2/huawei_solar #94: Integration stop working after update HA from 2.8.6 to 2.8.7

razvanu, 2022-08-26:
```
Logger: homeassistant.setup
Source: setup.py:184
First occurred: 4:21:04 PM (1 occurrences)
Last logged: 4:21:04 PM

Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.0.0rc6'].

```
Hi, Can you help with this error ?

Emilv2, 2022-08-26:
Some older versions have been deleted, but I don't really understand why you have this issue since the version in requirements is `huawei-solar>=1.2.0,<2.0.0`

razvanu, 2022-08-26:
I've deleted the folder from custom_components, and will try to reintegrate, but probably i've lost the energy logs.

razvanu, 2022-08-26:
And now it's gone completely from integration.

[image]

razvanu, 2022-08-26:
restored a backup, two weeks later , but still nothing

[image]

Emilv2, 2022-08-26:
Yes, huawei-solar 2.0.0rc6 has unfortunately been deleted, you won't be able to install that. Which version of the plugin do you have that it requires this version?

razvanu, 2022-08-26:
How to find plugin version ? Where to look ? 10x

Emilv2, 2022-08-26:
It's in manifest.json

razvanu, 2022-08-26:
```
  "domain": "huawei_solar",
  "name": "huawei_solar",
  "documentation": "https://github.com/Emilv2/huawei_solar",
  "dependencies": [],
  "codeowners": ["Emilv2"],
  "requirements": ["huawei-solar>=1.2.0,<2.0.0"],
  "version": "1.2.4"
}

```

wlcrs, 2022-08-27:
I have the impression that you had installed [my plugin](https://github.com/wlcrs/huawei_solar) earlier and have now tried to update by incorrectly getting the code from this repository. Please update to the latest version from my repository instead. Maybe consider using HACS for easy updating in the future?

razvanu, 2022-08-27:
I have HACS installed, now will reinstall

[image]

razvanu, 2022-08-29:
Solved, error on installation, you're right about this. Now i've installed from HACS. Thank you.

closed 2022-08-29 by razvanu