Emilv2/huawei_solar #96: Battery discharge seen as solar production.

stevenvw1973, 2022-10-11:
I have the Total_yield as source for solar production, and notice in the energy dashboard that battery discharge is also accounted in this value, resulting in the behavior that I have solar production even at night. You can see this in the screenshots below:
[image]

Is there another way of configurating this, so that the image is more correct ...

sven3, 2022-10-21:
I have noticed the same because I was confused that my solar panels deliver some 400 watts during night. I'm sorry I don't have a solution for this.

stevenvw1973, 2022-10-21:
I was able to make it work using this method: https://community.home-assistant.io/t/struggling-to-setup-energy-integration-for-huawei-solar-system-with-battery/367664

sven3, 2022-10-21:
Thanks for your hint, @stevenvw1973 . I tried to solve it with a sensor:

```
template:
  - sensor:
    - name: "Total"
      state: "{{ states('sensor.total_yield_hv1234567890') | float + states('sensor.storage_total_charge_hv1234567890') | float - states('sensor.storage_total_discharge_hv1234567890') | float }}"
      state_class: total_increasing
      unit_of_measurement: kWh
      device_class: energy

```
But it looks the real total yield is not ( total_yield + power into the battery - power from the battery ), because sometimes the value jumps back by some watts. Might also be a rounding issue.

After looking into the code it seems to me that @Emilv2 just passes the value unaltered, so it does not seem to be a bug in this project.

Emilv2, 2022-10-21:
This is an issue with the Huawei API, I can't do much about it. Here is a bit more about this issue: wlcrs/huawei_solar/issues/1

>But it looks the real total yield is not ( total_yield + power into the battery - power from the battery ), because sometimes the value jumps back by some watts. Might also be a rounding issue.

There are also conversion losses.