Emilv2/huawei_solar #101: Platform error sensor.huawei_solar - No module named 'pymodbus.client.asynchronous'

zsgodor, 2023-02-08:
Hi i update my HA Home Assistant 2023.2.3
20230202.0 - latest
My huawei_solar integration doesn't work, i see the following error.:
Platform error sensor.huawei_solar - No module named 'pymodbus.client.asynchronous' 
I reinstall the integration, but can't help.

Can you help me how can i solve this error.

Many thanks for your support.

I connect the inverter via wifi, and the connection is ok.:
ping 192.168.200.1
PING 192.168.200.1 (192.168.200.1) 56(84) bytes of data.
64 bytes from 192.168.200.1: icmp_seq=1 ttl=255 time=2.48 ms
64 bytes from 192.168.200.1: icmp_seq=2 ttl=255 time=3.34 ms
64 bytes from 192.168.200.1: icmp_seq=3 ttl=255 time=1.35 ms

wlx000f00acd96d: kapcsolódva to SUN2000-HV20C0067666 13
        "Ralink MT7601U"
        wifi (mt7601u), 00:0F:00:AC:D9:6D, hw, mtu 1500
        ip4 default
        inet4 192.168.200.100/24
        route4 0.0.0.0/0
        route4 192.168.200.0/24
        inet6 fe80::bb0c:c5e4:88fa:bf65/64
        route6 fe80::/64

nitruxa, 2023-02-14:
@zsgodor can you share your `configuration.yaml`? I'm having the same issue now.

zsgodor, 2023-02-14:
Hi @nitruxa 
I don't use conf.yaml, i use the modbus integration in HACS
Try add the integration Huawei modbus, enter your inverter ip address, try use port 502 or 6607.
I have installer, i connected directly the inverter, and enabled the modbus connection the inverter.
I think the new FW update make the problem.
[image]
[image]

nitruxa, 2023-02-14:
found this integration right before you wrote this. 
here is the reference to HACS custom component if someone will get here: https://github.com/wlcrs/huawei_solar