Emilv2/huawei_solar #102: pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.249', 502)

erikwachters, 2023-03-19:
Hi,
I use your plugin for more than a year now and had never a problem that I couldn't solve by rebooting HASS or re-download the integration.
But this morning suddenly it stopped working. I checked that there was a problem or update on my network but there wasn't anything. I still can check the data on the Fusion APP and I can ping to the invertor.
This is the warning in the log:
WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.249', 502)

Can you help or can you tell me how to fix it?

Thx

tomeeka, 2023-03-20:
Same problem for me, worked fine last week... (worked for years)
I am usint it through the internal wifi not the stick.
WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.200.1', 502)
Maybe huawei disabled the modbus connection remotely?

erikwachters, 2023-03-20:
It was yesterday at 0607 utc.
I use it also via the internal network.

tomeeka, 2023-03-20:
I had to reenable Modbus-TCP (unrestricted mode) in the Fusionsolar app with installer user on the inverter.
The port changed to 6607 from 502.
I rolled back HA to 2023.3.4 from the latest version.
There are problems with modbus for me with that release. ("No module named 'pymodbus.client.asynchronous'")
Now it is working...
The other huawei_solar integration (https://github.com/wlcrs) is not working for me. (could not connect to the inverter)

larrelandin, 2023-03-26:
> I had to reenable Modbus-TCP (unrestricted mode) in the Fusionsolar app with installer user on the inverter.
> The port changed to 6607 from 502.
> I rolled back HA to 2023.3.4 from the latest version. [...]

I seem to have the same issue as the rest. How do you get installer access to the app? I seem to have nowhere where I can enable modbus-TCP using my normal credentials.

tomeeka, 2023-03-26:
> I seem to have the same issue as the rest. How do you get installer access to the app? I seem to have nowhere where I can enable modbus-TCP using my normal credentials.

Sorry, not the Fusionsolar app. Use the SUN2000 app.
You have to connect to the internal wifi. Login with installer user. Default password: 00000a

larrelandin, 2023-03-28:
> Sorry, not the Fusionsolar app. Use the SUN2000 app. You have to connect to the internal wifi. Login with installer user. Default password: 00000a

Thanks for the clarification, however I can still not make it work. I have downgraded HA, set Modbus in unrestricted mode and added port number. Tried to upgrade HA as well after this but still no change.  All I get in the log is: "Failed to connect: [Errno 113] Connect call failed ('192.168.200.1', 6607)"
Tried to redo it all on a new installation but I still get the same error.
I have also made sure that my HA is attached to the SUN2000 WiFi but I cannot remember if there was any setting directly on the RPi or in the module for routing the request via the WiFi instead of the "normal" cable connection.

luisdomg, 2023-03-28:
I have the same problem. I did a scan from port 1 to 65534, it reports every port closed, which is bizarre as the app is working. I guess it's something about that restricted mode?