Emilv2/huawei_solar #104: Exception importing component in Home Assistant 2023.6

gerliczky, 2023-06-09:
I got the following exception, when I try to use the component

```
2023-06-09 05:18:38.640 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.huawei_solar.sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 833, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 850, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "", line 1206, in _gcd_import
  File "", line 1178, in _find_and_load
  File "", line 1149, in _find_and_load_unlocked
  File "", line 690, in _load_unlocked
  File "", line 940, in exec_module
  File "", line 241, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/sensor.py", line 26, in 
    from huawei_solar import AsyncHuaweiSolar, ConnectionException, ReadException
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in 
    from .huawei_solar import *  # noqa: F401 F403
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 15, in 
    from pymodbus.client.asynchronous.async_io import init_tcp_client
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 228, in 
    class ReconnectingAsyncioModbusTcpClient(object):
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 262, in ReconnectingAsyncioModbusTcpClient
    @asyncio.coroutine
     ^^^^^^^^^^^^^^^^^
AttributeError: module 'asyncio' has no attribute 'coroutine'
```

Home Assistant 2023.6.1
Supervisor 2023.06.1
Frontend 20230607.0 - latest

butsjoh, 2023-06-09:
I also had issue try to upgrade to 2023.6.0 yesterday. Something related to wrong yaml configuration or something. Not sure what i need todo to fix.

gm27271, 2023-06-11:
I have same error After upgrade to
Home Assistant 2023.6.1
Supervisor 2023.06.1
Operating System 10.2
Interfejs użytkownika: 20230608.0 - latest

111lisu, 2023-06-12:
I have same error After upgrade to 
Home Assistant 2023.6.1
Supervisor 2023.06.1
Operating System 10.2
Interfejs użytkownika: 20230608.0 - latest

Platform error sensor.huawei_solar - Exception importing custom_components.huawei_solar.sensor

gm27271, 2023-06-12:
If no feedback, then I Will consier going back to prvevious HA core version.

butsjoh, 2023-06-12:
I had to revert back to get it working again. I have no clue what is going on. Probably the python bump to 3.11 is what is causing the issue but i am not an expert at this. It would a least be great if somebody can confirm that that would be the issue.

butsjoh, 2023-06-12:
So i can confirm it is related to the python bump: https://stackoverflow.com/questions/74345065/attributeerror-module-asyncio-has-no-attribute-coroutine-in-python-3-11

butsjoh, 2023-06-12:
@Emilv2 any fix that can be applied for this?

titidnh, 2023-06-12:
You can upgrade the py package to a more recent version (manifest update)

{
  "domain": "huawei_solar",
  "name": "huawei_solar",
  "documentation": "https://github.com/Emilv2/huawei_solar",
  "dependencies": [],
  "codeowners": ["Emilv2"],
  "requirements": ["huawei-solar>=2.0.0,<=2.2.6"],
  "version": "1.2.4"
}

+ change the sensor.py file to use the new way to instantiate the class

        inverter = await AsyncHuaweiSolar.create(
            host=config[CONF_HOST], port=config[CONF_PORT], slave=config[CONF_SLAVE]
        )

It works back on my HA after those changes

Emilv2, 2023-06-12:
@titidnh Thanks! I currently can't test it, if someone else confirms it works I will push it as a fix.

butsjoh, 2023-06-13:
I can confirm it works as well ! thnx

agmuz, 2023-06-13:
@titidnh  hi i change in sensor.py line 169 and 170 , dont work
    try:
        inverter = await AsyncHuaweiSolar.create(
            host=config[CONF_HOST], port=config[CONF_PORT], slave=config[CONF_SLAVE]
        )

could you paste the whole sensor.py file

nalbor, 2023-06-13:
I've made those changes in manifest and sensor.py (line 169-170) and it didn't start working.

titidnh, 2023-06-13:
import asyncio
import logging
from datetime import timedelta

import homeassistant.helpers.config_validation as cv
import voluptuous as vol
from homeassistant.components.sensor import (
    ATTR_LAST_RESET,
    ATTR_STATE_CLASS,
    PLATFORM_SCHEMA,
    STATE_CLASS_MEASUREMENT,
    STATE_CLASS_TOTAL_INCREASING,
)
from homeassistant.const import (
    CONF_HOST,
    DEVICE_CLASS_CURRENT,
    DEVICE_CLASS_ENERGY,
    DEVICE_CLASS_POWER,
    DEVICE_CLASS_VOLTAGE,
    ENERGY_KILO_WATT_HOUR,
    POWER_WATT,
)
from homeassistant.helpers.entity import Entity
from homeassistant.util.dt import utc_from_timestamp

from huawei_solar import AsyncHuaweiSolar, ConnectionException, ReadException

_LOGGER = logging.getLogger(__name__)

SCAN_INTERVAL = timedelta(seconds=60)

# don't overload the poor thing
DEFAULT_COOLDOWN_INTERVAL = 0.1
# throttle reconnect attempts
DEFAULT_RECONNECT_INTERVAL = 30

CONF_OPTIMIZERS = "optimizers"
CONF_BATTERY = "battery"
CONF_SLAVE = "slave"
CONF_PORT = "port"

STATE_REGISTER = "active_power"

ATTR_GRID_CODE = "grid_code"
ATTR_SERIAL_NUMBER = "serial_number"
ATTR_NB_PV_STRINGS = "nb_pv_strings"

STATIC_ATTR_LIST = [
    "model_id",
    "model_name",
    ATTR_SERIAL_NUMBER,
    "rated_power",
    ATTR_NB_PV_STRINGS,
]

STATIC_ATTR_GRID_LIST = [
    "grid_standard",
    "grid_country",
]

DYNAMIC_ATTR_LIST = [
    "day_active_power_peak",
    "reactive_power",
    "power_factor",
    "efficiency",
    "grid_frequency",
    "grid_voltage",
    "grid_current",
    "line_voltage_A_B",
    "line_voltage_B_C",
[... 422 lines cut ...]
    @property
    def device_class(self):
        return self._device_class

    @property
    def available(self):
        """Return True if entity is available."""
        return self._parent_sensor.available

class HuaweiSolarDailyYieldSensor(HuaweiSolarEntitySensor):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)

    @property
    def extra_state_attributes(self):
        system_time = self._parent_sensor._attributes.get("system_time", None)
        if system_time is not None:
            last_reset = system_time.replace(hour=0, minute=0, second=0)
        else:
            last_reset = None

        return {
            **super().extra_state_attributes,
            ATTR_LAST_RESET: last_reset,
        }

titidnh, 2023-06-13:
[sensor.txt](https://github.com/Emilv2/huawei_solar/files/11734311/sensor.txt)

111lisu, 2023-06-13:
@agmuz  Ja przełożyłem na https://github.com/wlcrs/huawei_solar/ i śmiga aż miło i to bez grzebania w konfigu.

testgit1981, 2023-06-18:
Worked for me too

commit 57fd7e6a (2023-06-18): update huawei-solar requirements

Required for HA 2023.6 woth Python 3.11
Fixes #104
closed 2023-06-18 by Emilv2