Emilv2/huawei_solar #105: Huawei_solar stopped working after update of HA core from 2023.5.4 to 2023.6.2

pjgrun, 2023-06-20:
Huawei_solar stopped working after update of HA core from 2023.5.4 to 2023.6.2.
I was running version a973aed. tried to install version 57fd7e6, but HA reported error when checking configuration. Error message reported was:  platform error sensor.huawei_solar - exception importing custom_components.huawei_solar.sensor.
The only way to recover was to rollback HA core to version 2023.5.4
Is this a known issue?

garzew2, 2023-06-21:
I had same situation. I had to restore configuration from backup. After that update goes with any problem.
But after update Huawei_solar to new version it stop to work. I make test on HA 2023.5.4 and 2023.6.2.

garzew2, 2023-06-21:
This is my log when i update Huawei_solar 

`Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:181
Integration: Sensor (documentation, issues)
First occurred: 22:34:59 (1 occurrences)
Last logged: 22:34:59

Error while setting up huawei_solar platform for sensor
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/huawei_solar/registers.py", line 92, in decode
result = self.unit[result]
~~~~~~~~~^^^^^^^^
KeyError: 303

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 320, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/huawei_solar/sensor.py", line 181, in async_setup_platform
tmp = (await inverter.get(register)).value
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 215, in get
return (await self.get_multiple([name], slave))[0]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 254, in get_multiple
result = [await self._decode_response(registers[0], decoder)]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 207, in _decode_response
result = reg.decode(decoder, self)
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/huawei_solar/registers.py", line 94, in decode
raise DecodeError from err
huawei_solar.exceptions.DecodeError`

pjgrun, 2023-06-21:
I am now stuck on version 2023.5.4 version of HA core. With version 2023.6.2 of HA core, both old and new version of Huawei_Solar fail. Only possibility is to stay on HA core 2023.5.4.

Jarobwo, 2023-06-24:
I have the same problem with the new version, can you fix it?  do i need to look for another integration?

sdlafferty, 2023-06-25:
Same issue for me too.

EmAnzi3, 2023-06-25:
Same for me.

Trafal125, 2023-06-25:
Hi, same issue for me too. Any fix this problem?

Stea73, 2023-06-26:
I had the same issue. I restored a backup and avoided to update the addon. So far it works. Let's hope a fix will be implemented soon

Trafal125, 2023-06-26:
Hi, I update addons and I want get previous version - an older version. Anyone know how to download?

Stea73, 2023-06-26:
Not an expert, but either you download the previous version in github and replace the files manually (but I'm not sure it will work) or you restore a full backup prior to the addon update (you may lose any change done after the update though).

Trafal125, 2023-06-26:
I don't have backup this addon :(, and  I can't found option how get oldest version from github. Any sugest how to get old version this addon?

pjgrun, 2023-07-03:
Any news? Still blocked with this issue and thus unable to update HA core beyond 2023.5.4.

EmAnzi3, 2023-07-08:
I'm still stuck with this add-on not loaded.
I don't know if this might help but also from my mobile app FusionSolar I wasn't able to monitor my PV system. I addressed the issue changing inverter domain from intl.fusionsolar.huawei.com to region04eu5.fusionsolar.huawei.com

pjgrun, 2023-08-16:
Finally migrated to https://github.com/wlcrs/huawei_solar and adapted my application accordingly. The new integration is working fine with latest OS and HA Core versions, and at least in my case, is compatible with emilv2 integration except for the sensor naming. The new integration has additional features, like exposure of a number of services, besides the sensors. Happy to have finally found a solution.