Emilv2/huawei_solar #106: Error while setting up huawei_solar platform for sensor

jlopez77, 2023-07-11:
Since today (JUL-11-2023) at around 10:00AM the integration has stopped working no matter what I try (reinstall, reconfigure...)

This is the log output.

Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:169
Integration: Sensor ([documentation](https://next.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 1:25:27 PM (1 occurrences)
Last logged: 1:25:27 PM

Error while setting up huawei_solar platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 304, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/huawei_solar/sensor.py", line 169, in async_setup_platform
    inverter = AsyncHuaweiSolar(
TypeError: AsyncHuaweiSolar.__init__() got an unexpected keyword argument 'host'

pjgrun, 2023-07-11:
What version of HA Core do you have installed? Have you recently updated it?

jlopez77, 2023-07-11:
2023.04

No updates in months

bencer, 2023-07-31:
Use https://github.com/wlcrs/huawei_solar